riscv-software-src/riscv-tools #347: riscv-tools build.sh failed

athurwo, 2021-08-18:
Hi, failed log like blow:

/usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/fso/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: multiple definition of `bitbang_swd'; src/.libs/libopenocd.a(libocdjtagdrivers_la-remote_bitbang.o):/home/fso/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: first defined here
collect2: error: ld returned 1 exit status
gmake[2]: *** [Makefile:2897: src/openocd] Error 1
gmake[1]: *** [Makefile:4181: all-recursive] Error 1
gmake: *** [Makefile:1953: all] Error 2

Thanks

jim-wilson, 2021-08-23:
riscv-tools is unmaintained and stopped building a while ago.  You might want to avoid using it.  If you do want to use it, you should update the submodules as they are all 2+ years old.  Just cd into each subdir, git checkout master, and then git pull, to get recent copies of them.  If you want to double check, you can use git log -1 before and after to see the last commit.

jim-wilson, 2021-10-21:
There is a build log in the riscv-openocd dir you can try looking at.  But in general, riscv-tools isn't expected to build anymore because it has been unmaintained for too long.