riscv-software-src/riscv-tools #351: Always ending with error while building

sreersr, 2021-11-10:
usr/bin/ld: src/.libs/libopenocd.a(libocdjtagdrivers_la-bitbang.o):/home/sree/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: multiple definition of `bitbang_swd'; src/.libs/libopenocd.a(libocdjtagdrivers_la-remote_bitbang.o):/home/sree/riscv-tools/riscv-openocd/build/../src/jtag/drivers/bitbang.h:60: first defined here
collect2: error: ld returned 1 exit status
gmake[2]: *** [Makefile:2899: src/openocd] Error 1
gmake[1]: *** [Makefile:4183: all-recursive] Error 1
gmake: *** [Makefile:1955: all] Error 2

Hi I always get the above error, whenever build the tool. Kindly help me to fix this, my current dir is \home\sree\riscv-tools , what should be my path in **export RISCV=/path/to/install/riscv/toolchain**. Also in which dir Ariane design is to be placed. Thanks in advance.

jim-wilson, 2021-11-10:
riscv-tools doesn't build, and isn't being maintained.  The submodules are all about 3 years out of date.  This particular problem is fixed in the submodules.  If you update submodules you can fix this.  Just cd into each dir and do "git checkout master; git pull"  Except in riscv-openocd where you need to do "git checkout riscv; git pull".

sreersr, 2021-11-12:
Hi Jim,
Thanks for your email.
Still not working, screenshot is attached for your reference.

Thanks in advance.

With Best Regards,
*Sree*
Dr. Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"All is well"*

jim-wilson, 2021-11-12:
I don't see a screenshot.  If you sent email, try visiting the github link and adding the screenshot there.

sreersr, 2021-11-12:
Hi Jim,
Here you go!
[image]

jim-wilson, 2021-11-12:
I think that your copy of riscv-openocd is not up to date.  Try "git log | more" to see what version you have.  It should be November 12 2021.  Make sure that you are on the riscv branch and that you did a git pull to get the latest version.

jim-wilson, 2021-11-12:
If you do a git pull in the top level dir, it will revert changes to the submodules, so don't do that if you are updating submodules.

sreersr, 2021-11-13:
Hi Jim,
Yes it is Sep 12 2018, how to fix this. Please help me in this regard

sreersr, 2021-11-13:
Once I update riscv-openocd it is showing updated date as Nov 3 2021. but when I build riscv-tools I am getting below error:
oem@sree:~/Desktop/riscv-tools$ export RISCV=/home/oem/Desktop/riscv-tools
oem@sree:~/Desktop/riscv-tools$ ./build.sh 
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
configure: error: Internal libjaylink not found, run either 'git submodule init' and 'git submodule update' or disable internal libjaylink with --disable-internal-libjaylink.
oem@sree:~/Desktop/riscv-tools$

sreersr, 2021-11-13:
Please review:

oem@sree:~/Desktop/riscv-tools$ cd riscv-openocd/
oem@sree:~/Desktop/riscv-tools/riscv-openocd$ git log | more
commit f4f8b59f620725ca437c4d8aab571538f2278e13
Author: Tim Newsome <tim@sifive.com>
Date:   Fri Nov 12 11:29:00 2021 -0800

    Properly save/restore vtype.ill (#661)
    
    Change-Id: I2478be8a849ceb4f637bbcfb774099217c509dfd
    Signed-off-by: Tim Newsome <tim@sifive.com>

commit 641e51ff7fd0e73c8449e2994a55b64674ef17d6
Author: Jan Matyas <50193733+JanMatCodasip@users.noreply.github.com>
Date:   Thu Nov 4 17:48:23 2021 +0100

    Regenerated debug_defines.h and encoding.h (#659)
    
    The main intention is to get access to some of the CSRs
    that were so far unknown to OpenOCD (tinfo, mcountinhibit, ...).

commit bda1027191cfc49a6627d52cd6e894b765afd30b
Merge: d97da0eb7 b1f244b82
Author: Tim Newsome <tim@sifive.com>
Date:   Wed Nov 3 11:16:55 2021 -0700

    Merge pull request #660 from riscv/resume_order
    
--More--
[6]+  Stopped                 git log | more
oem@sree:~/Desktop/riscv-tools/riscv-openocd$ cd ..
oem@sree:~/Desktop/riscv-tools$ export RISCV=/home/oem/Desktop/riscv-tools
oem@sree:~/Desktop/riscv-tools$ ./build.sh 
Starting RISC-V Toolchain build process

libtoolize: putting auxiliary files in AC_CONFIG_AUX_DIR, 'build-aux'.
libtoolize: copying file 'build-aux/ltmain.sh'
libtoolize: Consider adding 'AC_CONFIG_MACRO_DIRS([m4])' to configure.ac,
libtoolize: and rerunning libtoolize and aclocal.
libtoolize: Consider adding '-I m4' to ACLOCAL_AMFLAGS in Makefile.am.
libtoolize: 'AC_PROG_RANLIB' is rendered obsolete by 'LT_INIT'
configure.ac:22: installing 'build-aux/compile'
configure.ac:34: installing 'build-aux/config.guess'
configure.ac:34: installing 'build-aux/config.sub'
configure.ac:17: installing 'build-aux/install-sh'
configure.ac:17: installing 'build-aux/missing'
Makefile.am: installing './INSTALL'
Makefile.am: installing 'build-aux/depcomp'
Makefile.am:24: installing 'build-aux/mdate-sh'
Makefile.am:24: installing 'build-aux/texinfo.tex'
Configuring project riscv-openocd
configure: error: jimtcl not found, run git submodule init and git submodule update.
oem@sree:~/Desktop/riscv-tools$ cd riscv-openocd/
oem@sree:~/Desktop/riscv-tools/riscv-openocd$ git log | more
commit f4f8b59f620725ca437c4d8aab571538f2278e13
Author: Tim Newsome <tim@sifive.com>
Date:   Fri Nov 12 11:29:00 2021 -0800

    Properly save/restore vtype.ill (#661)
    
    Change-Id: I2478be8a849ceb4f637bbcfb774099217c509dfd
    Signed-off-by: Tim Newsome <tim@sifive.com>

commit 641e51ff7fd0e73c8449e2994a55b64674ef17d6
Author: Jan Matyas <50193733+JanMatCodasip@users.noreply.github.com>
Date:   Thu Nov 4 17:48:23 2021 +0100

    Regenerated debug_defines.h and encoding.h (#659)
    
    The main intention is to get access to some of the CSRs
    that were so far unknown to OpenOCD (tinfo, mcountinhibit, ...).

commit bda1027191cfc49a6627d52cd6e894b765afd30b
Merge: d97da0eb7 b1f244b82
Author: Tim Newsome <tim@sifive.com>
Date:   Wed Nov 3 11:16:55 2021 -0700

    Merge pull request #660 from riscv/resume_order
    
--More--
[7]+  Stopped                 git log | more
oem@sree:~/Desktop/riscv-tools/riscv-openocd$ git submodule init 
Submodule 'jimtcl' (https://github.com/msteveb/jimtcl.git) registered for path 'jimtcl'
Submodule 'src/jtag/drivers/libjaylink' (https://repo.or.cz/libjaylink.git) registered for path 'src/jtag/drivers/libjaylink'
Submodule 'tools/git2cl' (https://repo.or.cz/git2cl.git) registered for path 'tools/git2cl'
oem@sree:~/Desktop/riscv-tools/riscv-openocd$ git submodule update 
Cloning into '/home/oem/Desktop/riscv-tools/riscv-openocd/jimtcl'...
Cloning into '/home/oem/Desktop/riscv-tools/riscv-openocd/src/jtag/drivers/libjaylink'...
Cloning into '/home/oem/Desktop/riscv-tools/riscv-openocd/tools/git2cl'...
Submodule path 'jimtcl': checked out '2d66360c61d2a89d4008e8bad12ae3aa5f0331e2'
Submodule path 'src/jtag/drivers/libjaylink': checked out '9aa7a5957c07bb6e862fc1a6d3153d109c7407e4'
Submodule path 'tools/git2cl': checked out '8373c9f74993e218a08819cbcdbab3f3564bbeba'
oem@sree:~/Desktop/riscv-tools/riscv-openocd$ cd ..
oem@sree:~/Desktop/riscv-tools$ export RISCV=/home/oem/Desktop/riscv-tools
oem@sree:~/Desktop/riscv-tools$ ./build.sh 
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
../../../../../src/jtag/drivers/libjaylink/configure.gnu: line 2: /home/oem/Desktop/riscv-tools/riscv-openocd/build/src/jtag/drivers/libjaylink/../../../../../src/jtag/drivers/libjaylink/configure: No such file or directory
configure: error: ../../../../../src/jtag/drivers/libjaylink/configure.gnu failed for src/jtag/drivers/libjaylink
oem@sree:~/Desktop/riscv-tools$

jim-wilson, 2021-11-14:
Unfortunately riscv-tools is a mess because no one is maintaining it.  It might be easier to compile the individual tools separately.

It sounds like the riscv-openocd subdir is not OK.  Try running "git status".  There might be submodules that are missing, or submodules that aren't supposed to be there.  Also try "git branch" to make sure you are on the riscv branch.

Checking your log messages, the first one tells you what the problem is and how to fix:
Configuring project riscv-openocd
configure: error: jimtcl not found, run git submodule init and git submodule update.

sreersr, 2021-11-16:
Hi Jim,
Thanks for your insights.

I fixed open riscv-openocd issues with your suggestion. Now while running
make verilate in ariane , I am getting below error:

/bin/sh: line 1: verilator: command not found
make: *** [Makefile:659: verilate] Error 127

I also installed Verilator as per suggestion, but still getting error. can
you help me how to include verilator in the ariane-sdk. Also I am bit
confused on which directories ariane, riscv-tools and verilator should be
installed, please advise. Thanks in advance.

With Best Regards,
*Sree*
Dr. Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"All is well"*

jim-wilson, 2021-11-16:
I'm not a hardware guy, and don't know anything about running ariane on verilator.  You should proabably ask the ariane folks that question.

sreersr, 2021-11-16:
Thanks, I'll check it out.
With Best Regards,
*Sree*
Dr. Sree Ranjani Rajendran,
FICS Research Lab, University of Florida, Gainesville.
*******************************************************
*"All is well"*

sreersr, 2021-11-17:
Hi Jim,
Now I am getting the below error, please help me in this regard:
oem@sree:~/Desktop/riscv-tools$ export RISCV=/home/oem/Desktop/riscv-tools/
oem@sree:~/Desktop/riscv-tools$ ./build.sh 
Starting RISC-V Toolchain build process

Removing existing riscv-openocd/build directory
Configuring project riscv-openocd
Building project riscv-openocd
Installing project riscv-openocd

Removing existing riscv-isa-sim/build directory
Configuring project riscv-isa-sim
Building project riscv-isa-sim
../fesvr/dtm.cc: In member function ‘uint32_t dtm_t::get_xlen()’:
../fesvr/dtm.cc:488:16: error: ‘runtime_error’ is not a member of ‘std’
  488 |     throw std::runtime_error("FESVR DTM Does not support 128-bit");
      |                ^~~~~~~~~~~~~
../fesvr/dtm.cc:505:14: error: ‘runtime_error’ is not a member of ‘std’
  505 |   throw std::runtime_error("FESVR DTM can't determine XLEN. Aborting");
      |              ^~~~~~~~~~~~~
../fesvr/dtm.cc:506:1: warning: control reaches end of non-void function [-Wreturn-type]
  506 | }
      | ^
gmake: *** [Makefile:313: dtm.o] Error 1

sreersr, 2021-11-17:
Hi Jim, 
I have the below error while building riscv-tools, please help me in this regard

Removing existing riscv-pk/build directory
Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small
gmake: *** [Makefile:336: file.o] Error 1

jim-wilson, 2021-11-18:
Do you have a RISC-V compiler?  The errors look like x86_64 compiler errors.  You need to have a riscv64-unknown-elf-gcc (usually from riscv-gnu-toolchain) in your path, or the makefile will try to use the native compiler with RISC-V command line arguments and you get an error like this.

sreersr, 2021-11-18:
Dear Jim,
Thanks for your reply.
I think I am missing something like adding a right path for tool installation. Please help me on the below naive questions :
I am choosing to install all tools (riscv-tools and riscv-gnu-toolchain) in /home/usr/Desktop/riscv-tools.
so how to fix my path and where to add my path? is it in bashrc file or just export in directory I am working on. 

Thanks in advance. 
**Installation (Linux)
To build the Linux cross-compiler, pick an install path. If you choose, say, /opt/riscv, then add /opt/riscv/bin to your PATH now. Then, simply run the following command:

./configure --prefix=/opt/riscv
make linux**

jim-wilson, 2021-11-18:
build.common does this
```
PATH="$RISCV/bin:$PATH"
```
so you need to remove the /bin from your RISCV environment variable.  The value for RISCV needs to be the same value you passed to the --prefix environment variable.