riscv-software-src/riscv-tools #353: Build failure OS X: illegal options for BSD ar

ppannuto, 2022-02-18:
Trying to build on OS X fails around `make: *** [libfesvr.a] Error 1`. Digging into this a bit, it is because the build relies on GNU-only `ar` options:

```bash
[-bash] Fri 18 Feb 10:59 [[master] ~/code/external/riscv-tools]
$ cd riscv-isa-sim/build/
/Users/ppannuto/code/external/riscv-tools/riscv-isa-sim/build

[-bash] Fri 18 Feb 10:59 [[(v1.0.0)] ~/code/external/riscv-tools/riscv-isa-sim/build]
$ ar rcs -o libfesvr.a elfloader.o htif.o memif.o dtm.o syscall.o device.o rfb.o context.o htif_pthread.o htif_hexwriter.o dummy.o option_parser.o term.o tsi.o
ar: illegal option combination for -r
usage:  ar -d [-TLsv] archive file ...
	ar -m [-TLsv] archive file ...
	ar -m [-abiTLsv] position archive file ...
	ar -p [-TLsv] archive [file ...]
	ar -q [-cTLsv] archive file ...
	ar -r [-cuTLsv] archive file ...
	ar -r [-abciuTLsv] position archive file ...
	ar -t [-TLsv] archive [file ...]
	ar -x [-ouTLsv] archive [file ...]

[-bash] Fri 18 Feb 10:59 [[(v1.0.0)] ~/code/external/riscv-tools/riscv-isa-sim/build]
$ which ar
/usr/bin/ar

[-bash] Fri 18 Feb 10:59 [[(v1.0.0)] ~/code/external/riscv-tools/riscv-isa-sim/build]
$ /usr/local/opt/binutils/bin/ar rcs -o libfesvr.a elfloader.o htif.o memif.o dtm.o syscall.o device.o rfb.o context.o htif_pthread.o htif_hexwriter.o dummy.o option_parser.o term.o tsi.o
```

Where `/usr/bin/ar`, the default `ar` is the version installed natively on Mac, while:
```
$ /usr/local/opt/binutils/bin/ar --version
GNU ar (GNU Binutils) 2.37
Copyright (C) 2021 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
```
is the GNU version installed via `brew install binutils`.

ppannuto, 2022-02-18:
Update: mixing toolchains (i.e. using the GNU `ar` while otherwise using mac-native tools via `AR=/usr/local/opt/binutils/bin/ar ./build.sh`) results in more errors:

```
ld: warning: ignoring file libjim.a, building for macOS-x86_64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture x86_64:
  "_Jim_CreateInterp", referenced from:
      _main in jimsh.o
  "_Jim_Eval", referenced from:
      _main in jimsh.o
  "_Jim_EvalFile", referenced from:
      _main in jimsh.o
  "_Jim_EvalSource", referenced from:
      _Jim_initjimshInit in initjimsh.o
  "_Jim_FreeInterp", referenced from:
      _main in jimsh.o
  "_Jim_GetExitCode", referenced from:
      _main in jimsh.o
  "_Jim_InitStaticExtensions", referenced from:
      _main in jimsh.o
  "_Jim_InteractivePrompt", referenced from:
      _main in jimsh.o
  "_Jim_ListAppendElement", referenced from:
      _JimSetArgv in jimsh.o
  "_Jim_MakeErrorMessage", referenced from:
      _main in jimsh.o
  "_Jim_NewIntObj", referenced from:
      _main in jimsh.o
      _JimSetArgv in jimsh.o
  "_Jim_NewListObj", referenced from:
      _main in jimsh.o
      _JimSetArgv in jimsh.o
  "_Jim_NewStringObj", referenced from:
      _main in jimsh.o
      _JimSetArgv in jimsh.o
  "_Jim_PackageProvide", referenced from:
      _Jim_initjimshInit in initjimsh.o
  "_Jim_RegisterCoreCommands", referenced from:
      _main in jimsh.o
  "_Jim_SetVariableStr", referenced from:
      _main in jimsh.o
      _JimSetArgv in jimsh.o
  "_Jim_SetVariableStrWithStr", referenced from:
      _main in jimsh.o
  "_Jim_String", referenced from:
      _main in jimsh.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [jimsh] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

Some light googling supports that the 'building for macOS-x86_64 but linking for macOS-x86_64' is indeed a symptom of trying to mix toolchains.

jim-wilson, 2022-03-07:
The GNU toolchain has a lot of assumptions that it will be used with gnu tools, and riscv-gnu-toolchain can't fix this.  So yes, you need to use gnu tools as much as possible when building it.  Using homebrew is recommended.  I don't have access to a Mac, so I've never tested builds on Mac OS X.  You might try asking someplace like the FSF GCC gcc-help mailing list.

The SiFive freedom-tools github repo has support for building on Mac OS X.  But it only supports building a cross compiler for embedded elf so I don't know if that is helpful or not.
[https://github.com/sifive/freedom-tools](https://github.com/sifive/freedom-tools)