riscv-software-src/riscv-tools #354: REQUESTING ADVICE ON RISCV-OPCODES

BhattSoham, 2022-04-05:
Hi all,
     Good day. I am trying to run a code using riscv32-unknown-elf-gcc, but every time it is giving the same result as :
Assembler messages:
Error: internal: bad RISC-V opcode (bits 0xffffffffffff8f80 undefined): slti
Fatal error: internal: broken assembler.  No assembly attempted

But, in my code, there's no 'slti' opcode has been used.

My code is:
'
int mult() {
        int a=1000,b=3;
        return a*b;
}

int main() {
        mult();
}

'
I am trying to run the code using the following command:
riscv32-unknown-elf-gcc -g mul.c -o mul

If I am trying to run other codes, it is also giving the same error again and again.

I tried the following commands:
./configure --help | grep abi
./configure --prefix=/opt/riscv32/bin --with-arch=rv32im --with-abi=ilp32
sudo make clean
sudo make all
export PATH=/opt/riscv32/bin:$PATH
riscv32-unknown-elf-gcc -g mul.c -o mul

Your time and consideration is appreciated.

Thanks in advance!

Sincerely,
Soham.

jim-wilson, 2022-04-06:
Binutils doesn't use riscv-opcodes so I don't know why you are mentioning that.  Also, there is no copy of binutils in riscv-tools, so it is unclear why a binutils problem has been reported here.  There was a copy years ago.  Are you trying to use old and broken versions of riscv-tools?

The assembler error means that your copy of GNU as is broken.  Did you modify it?  If so, then your patches are wrong.  Otherwise, I don't know of any reason why binutils would be broken.

You need to start from the beginning, and report all info necessary to reproduce your bug.  Exactly what sources you are using, e.g. what git tree and git branch/label/commitid.  And what changes you have made to the sources.  And how you configured and built them.

There is a copy of binutils in riscv-gnu-toolchain.  The only supported version of riscv-gnu-toolchain is the top of tree.   Don't try using random branches of riscv-binutils-gdb; they are unlikely to work.

BhattSoham, 2022-04-06:
Thanks for the reply. Is there any complete link from where I can download the riscv-tools?

jim-wilson, 2022-04-06:
You download sources from here.  There are no binary releases as far as I know of.  Also, I should mention that riscv-tools hasn't been maintained for several years now, and may not build.  It is probably best to avoid it.  If you need an individual tool like riscv-isa-sim, then try cloning that directly instead of using riscv-tools.