riscv-software-src/riscv-tools #361: unrecognized argument in option ‘-mcmodel=medany’

ardeal, 2022-12-19:
Hi, 

On ubuntu 18.04, I am experience the following issue when I run `bash build.sh` in riscv-tools folder:

```C
Removing existing riscv-isa-sim/build directory
Configuring project riscv-isa-sim
Building project riscv-isa-sim
../spike_main/disasm.cc: In constructor ‘disassembler_t::disassembler_t(int)’:
../spike_main/disasm.cc:275:1: note: variable tracking size limit exceeded with -fvar-tracking-assignments, retrying without
 disassembler_t::disassembler_t(int xlen)
 ^~~~~~~~~~~~~~
Installing project riscv-isa-sim

Removing existing riscv-pk/build directory
Configuring project riscv-pk
Building project riscv-pk
gcc: error: unrecognized argument in option ‘-mcmodel=medany’
gcc: note: valid arguments to ‘-mcmodel=’ are: 32 kernel large medium small; did you mean ‘medium’?
gmake: *** [file.o] Error 1

```

aswaterman, 2022-12-19:
Seems very similar to https://github.com/riscv-software-src/riscv-pk/issues/204 — I can’t help further, but maybe you’ll find some inspiration there.

ardeal, 2022-12-20:
@aswaterman 
Thank you!
I just notice that I misunderstand risc-tools and riscv-gnu-toolchain.

I just now downloaded riscv-gnu-toolchain from: 
1) https://github.com/riscv-collab/riscv-gnu-toolchain/releases/download/2022.12.17/riscv64-elf-ubuntu-20.04-nightly-2022.12.17-nightly.tar.gz
2) https://github.com/riscvarchive/riscv-fesvr
3) https://github.com/RISCV-Tools/riscv-tools

then I tar the tool to /opt/riscv folder.

In my bashrc file, I added:
export RISCV="/opt/riscv"
export PATH=$PATH:$RISCV/bin
and run source ~/.bashrc

when I run `./build.sh` in riscv-tools folder, I encountered the following issue:

```C
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: dhrystone.riscv has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: spmv.riscv has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: mt-vvadd.riscv has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: mt-matmul.riscv has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: pmp.riscv has a LOAD segment with RWX permissions
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccsdE6df.o: in function `tohost':
(.tohost+0x0): multiple definition of `tohost'; /tmp/cchaqrog.o:(.sbss+0x10): first defined here
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: /tmp/ccsdE6df.o: in function `fromhost':
(.tohost+0x40): multiple definition of `fromhost'; /tmp/cchaqrog.o:(.sbss+0x8): first defined here
/opt/riscv/lib/gcc/riscv64-unknown-elf/12.2.0/../../../../riscv64-unknown-elf/bin/ld: warning: rv32ui-v-simple has a LOAD segment with RWX permissions
collect2: error: ld returned 1 exit status
make[1]: *** [/home/au/e203/riscv-tools/riscv-tests/build/../isa/Makefile:74: rv32ui-v-simple] Error 1
make: *** [Makefile:29: isa] Error 2

```